memristor/mep3 #44: Hendlovanje potencijalnog reseta kretacke ploce - mep3_driver

angstrem98, 2021-12-28:
Veoma je mala sansa, ali ipak je moguce da se desi restart kretacke ploce tokom rada.

lukicdarkoo, 2021-12-28:
Ovo nam nije palo na pamet kad smo razmisljali da sa kretacke saljemo poziciju ili brzinu. Slanjem brzine ne bismo imali problem sa resetom kretacke.

Ali, ako se kretacka resetuje onda imamo nekih bas ozbiljnih problema :)

angstrem98, 2021-12-28:
> Ovo nam nije palo na pamet kad smo razmisljali da sa kretacke saljemo poziciju ili brzinu. Slanjem brzine ne bismo imali problem sa resetom kretacke.
> 
> Ali, ako se kretacka resetuje onda imamo nekih bas ozbiljnih problema :)

Veci je problem ako saljemo brzinu. Ako saljemo brzinu i propustimo makar jedan paket imamo ozbiljan gubitak pozicije. 
Razmisljam o tome da kretacka kad se resetuje posalje na CAN poruku koja znaci da je resetovana (one shot), tad mep3_driver upisuje u registre enkodera poziciju koju je zapamtio (ima informaciju u prethodnom stanju enkodera) i nastavlja rad.

lukicdarkoo, 2021-12-29:
> propustimo makar jedan paket imamo ozbiljan gubitak pozicije.

To bismo rijesili tako sto bismo interpolirali brzinu.

> Moguce je da usled ukljucenja pumpi padne napon na DC medjukolu, ali moze se resiti lako dodavanjem kondenzatora.

Posao elektronicara je da se to ne desi :) Ja se ne sjecam da smo imali slicnih problema