memristor/mep3 #75: Internet pristup za robote

angstrem98, 2022-01-27:
Ovo treba hitno resiti, previse vremena mi oduzima, nekad treda samo jedan paket da instaliram i moram menjati hotspot-ove. Plus treba nam pristup git repozitorijumu.

angstrem98, 2022-01-27:
@filiparag Imas ideju kako je najbolje resiti?

filiparag, 2022-01-28:
> Imas ideju kako je najbolje resiti?

Ako uspemo da pristupimo onom OpenWRT ruteru problem ce se resiti brzo, a ovako jedino da utaknes taj AP switch u ethernet u G3 pa vidis da li ce da ozivi automatski.

filiparag, 2022-04-06:
Firmware: https://openwrt.org/toh/tp-link/tl-wdr3600_v1

VladimirVincan, 2022-04-07:
Našao sam dva rutera koja zadovoljavaju naše zahteve:
[TP-Link M7350](http://www.netiks.rs/4g-lte-mobile-wi-fi)
[Cudy LT500](http://www.netiks.rs/cudy-lt500-lte-ruter-sa-modemom-ac1200)
Kako vam se sviđaju?

filiparag, 2022-04-07:
Ovaj sa antenama (Cudy LT500) mi izgleda fino, ali mu treba spoljno napajanje pa nije toliko fleksibilan. A i TP-Link M7350 je dobar isto, al je i skuplji 🤑 Sve zavisi da li nam se daje toliko para za ruter 🤷🏻‍♂️

closed 2022-04-16 by filiparag