memristor/mep3 #79: mep3_navigation neće da se build-uje u main brenču

VladimirVincan, 2022-02-05:
Starting >>> mep3_navigation
--- stderr: mep3_navigation                                         
CMake Error at CMakeLists.txt:23 (find_package):
  By not providing "Findnav2_core.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "nav2_core", but CMake did not find one.

  Could not find a package configuration file provided by "nav2_core" with
  any of the following names:

    nav2_coreConfig.cmake
    nav2_core-config.cmake

  Add the installation prefix of "nav2_core" to CMAKE_PREFIX_PATH or set
  "nav2_core_DIR" to a directory containing one of the above files.  If
  "nav2_core" provides a separate development package or SDK, be sure it has
  been installed.

make: *** [Makefile:326: cmake_check_build_system] Error 1
__

VladimirVincan, 2022-02-05:
U pitanju je commit 1e4ca4ad3218528abc2a4589bdb2411fa7eeec11

angstrem98, 2022-02-07:
Nisam video Issue ranije. Meni radi bez problema, pustano i na robotu.
Probao da reinstaliras nav2 pakete?
```sudo apt install ros-galactic-navigation2```
```sudo apt install ros-galactic-nav2-core```

angstrem98, 2022-02-11:
Zatvaram issue, zastarelo.

closed 2022-02-11 by angstrem98