memristor/mep3 #86: PreciseNavigateToActionIssue

VladimirVincan, 2022-02-19:
@angstrem98 
Napravili smo prvu strategiju pomoću komande PreciseNavigateToAction, međutim ta akcija se vrti više nego što bi trebalo. Molim te uđi u commit e732ccdcfe61a8915fb4f6160ae00ea916ff8c98, i pokreni sledeće komande:

ros2 launch mep3_bringup simulation_launch.py

ros2 run mep3_behavior_tree mep3_behavior_tree first_strategy --ros-args -r __ns:=/big

https://user-images.githubusercontent.com/13640533/154808369-e659ffac-d8a0-44f7-852d-7b28b6bc5dc6.mp4

angstrem98, 2022-02-19:
Ne moram ni otvoriti, znam sta je problem. Webots objavljuje /odom na 50 Hz umesto na 100 Hz i to sve pobrljavi. Napravicu neki hack da se to resi.

closed 2022-02-24 by angstrem98

lukicdarkoo, 2022-03-10:
Sredjeno u [03feb23](https://github.com/memristor/mep3/pull/116/commits/03feb230174c16258e05dcd7ad64dfb8f85631c3)