memristor/mep3 #104: Bad rotation estimation with ArUco tags (z axis flipped)

VladimirVincan, 2022-03-03:
[image]

Additional info about the problem can be found on this [link](https://github.com/opencv/opencv/issues/8813).
We need to find a way to tell the PnP algorithm that the _z_ axis is always faced up.
Also, it would be nice if we could tell the algorithm that the _z_ axis distance from the world frame is always 510 mm, in order to minimize the error.

VladimirVincan, 2022-03-05:
Pogledati ovaj komentar i sledeći algoritam: [komentar](https://stackoverflow.com/questions/51709522/unstable-values-in-aruco-pose-estimation#comment90618214_51736875) [algoritam](https://docs.opencv.org/4.1.0/d9/d0c/group__calib3d.html#ga549c2075fac14829ff4a58bc931c033d)

VladimirVincan, 2022-03-05:
Examples of wrong orientations ([normalized vector], degrees):
angle-axis: [-0.33265696  0.46720817  0.81918   ], 141.78825742681315

angstrem98, 2022-03-11:
Šta misliš da ti kad detektujes aruco (dobiješ njegov frame) proveriš da li osa z gleda na gore i ako ne ručno primeniš dodatnu rotaciju na frame i onda objaviš? Malo primitivno rešenje, ali čini mi se dosta brže nego da se razvija algoritam dalje.

closed 2022-03-16 by VladimirVincan