memristor/mep3 #112: tf2 listener neće da učita statičke transformacije

VladimirVincan, 2022-03-06:
Imam problem sa učitavanjem statičkih transformacija terena u  _robot detection node_, ako pokrenem:
`ros2 run mep3_localization static_broadcaster`
`ros2 run mep3_localization robot_detection`
Trebalo bi da se transformacije između kamere, mape i markera 42 učitaju u tf2 stablo, što se i vizualizuje u RViz-u i prikazuje preko komande npr. `ros2 run tf2_ros tf2_echo map camera_static`, kao i da se učitaju u  _robot detection node_, ali se kod ovde i dalje blokira:

https://github.com/memristor/mep3/blob/afc3e842d4f10f7c62caa9898b156e3399a71fb6/mep3_localization/mep3_localization/robot_detection.py#L70-L74

Nije nešto suštinski bitno, mogu se uneti transformacije ručno u kod, ali mi ovo deluje kao elegantnije rešenje.

angstrem98, 2022-03-07:
Probaj sa ovim remapiranjem:
https://github.com/memristor/mep3/blob/main/mep3_bringup/launch/robot_launch.py#L91

prakticno `/tf_static` treba remapirati na `/big/tf_static`

angstrem98, 2022-03-09:
Zatvaram issue jer gore navedeno remapiranje rešava problem.

closed 2022-03-09 by angstrem98