memristor/mep3 #121: Prečice za terminal

filiparag, 2022-03-12:
Za brži rad u terminalu predlažem da dodamo sledeće *alias*-e u `~/.bashrc` razvojnog okruženja:
- **`cb`** `colcon build --symlink-install`
- **`rr`** `rm -rf ~/ros2_ws/install/ ~/ros2_ws/build/`
- **`s`** `source ~/ros2_ws/install/local_setup.bash`
- **`sim`** `ros2 launch mep3_simulation simulation_launch.py`
- **`br`** `ros2 launch mep3_bringup simulation_launch.py`
- **`rv`** `ros2 launch mep3_bringup rviz_launch.py`

Ako imate još neku ideju, pišite u komentarima 🙃

lukicdarkoo, 2022-03-13:
Ja bih ovo pretvorio u tabelu, kolone da budu [komanda, opis, alias] i dodao kao sekciju u README pod nazivom Cheatsheet.

Dodao bih jos komande za skoro sve BT cvorove (i da se mogu proslijediti argumenti kao u BT cvorovima) koje imamo, recimo navigate to pose, precise navigate to pose...

angstrem98, 2022-03-13:
Moze li to da se stavi sve u jedan fajl pa da u `~./bashrc` source-ujemo njega, da nemamo vise tih linija?

VladimirVincan, 2022-03-14:
Dodao bih još:

- `we` `~/ros2_ws/src/mep3/mep3_simulation/webots_data/worlds/eurobot_2022.wbt `